scooterhacking/scooterhacking-issues-public #127: BUG: little working e-brake on max G2

rok-medic, 2024-10-09:
### Scooterhacking Tool

Firmware (SHFW)

### Type of issue

Performance issue

### Severity of issue

high (major functionality broken, no workaround)

### SHUT Version

SHUT 2.6.1

### SHFW Version

SHFW Base170 3.9.1

### Scooter model

Ninebot G2

### Description of the Bug

Installed SHFW 3.9.1 and I set only throttle, break, field weakening wheel size and never touched other.

but e break only works under 10km/h.
it doesn't work upper 10km/h.
and autobreak works no issues

My tested
1. break setting
graph : 55A Flat curve
overshoot current : 65A
boost : 100%
auto breaking : off

It doesn't work

2. graph : Advanced 1 52000, Advanced 2 53000, Advanced 3 54000, Advanced 4 55000
overshoot current : 65A
boost : 100%
auto breaking : off

It doesn't work

and i never understand this why "only works under10km/h?"
so i tested ofw 1.4.8 40km and ofw 1.7.3 it works good!

this ploblem works only shfw and it means its not a hardware issues!
i wanna use shfw but it makes big dangers....

i upload my setting capture and my segway official app on dashboard

i set the other ofw and there's work nomally. and i reinstall shfw doesn't work again.

[image]

https://github.com/user-attachments/assets/190504d8-9715-42c1-afb0-5b5218f504df

### Steps to Reproduce

https://github.com/user-attachments/assets/e01140fc-b323-4ea6-b02f-45579175a48b

[image]

setting like this capture and riding

upper 10km/h a little works e break like video andonly pysical breaks work
and under 10km/h works e break

the video is segway official app dashboard, break w is never go up around 100 w
break lights are nomal work while i use break,
its not a hardware issues

### Expected vs. Actual Behavior

all time i use break, e breake and pysical break both working
but upper 10km/h e break doesnt work only pysical break work
under 10km/h both break nomally work

### Contact Information

giant0163

lekrsu, 2024-10-09:
The e-brake behavior is already changed and being worked on for pre-release.

closed 2024-10-09 by lekrsu

rok-medic, 2024-10-09:
I dont konw what u says. whats the change thing? my break doesnt work, not functional ploblem

rok-medic, 2024-10-09:
> e-브레이크 동작은 이미 변경되어 사전 출시를 위해 작업 중입니다.

u mean developers are fixing bug on e break?

rok-medic, 2024-10-09:
> e-브레이크 동작은 이미 변경되어 사전 출시를 위해 작업 중입니다.

my e break is real whell power by reverse motor break
u mean this?