scooterhacking/scooterhacking-issues-public #128: BUG: Broken profile triggers on Mi Scooter Pro

Samybigio, 2024-10-09:
### Scooterhacking Tool

Firmware (SHFW)

### Type of issue

Other

### Severity of issue

medium (affects usability but has a workaround)

### SHUT Version

2.6.1

### SHFW Version

3.9.1

### Scooter model

Xiaomi Pro, Pro2, 1s, essential, 3

### Description of the Bug

profile triggers will switch profile even if brake isnt pressed (example: brake and single power button press will activate despite brake not being pressed). this means that turning the headlight on/off or changing speed mode (eco, drive and sport) to trigger profile change by accident. this only happens on my Mi scooter Pro, the functionality works as expected in my Scooter Pro 2. this happened on older versions of SHFW as well. oh and another thing: this happens on "hold brake and power button press" and "hold brake and double power button press", for some reason "hold brake and hold throttle" works as expected since holding accelerator by itself wont do anything.

### Steps to Reproduce

1. set brake + single power button press profile trigger to switch to profile 2
2. when turning on the scooter, switching on the headlight works fine but once you try to turn it off it will trigger a profile change (from profile 1 to 2 in this case)

### Expected vs. Actual Behavior

i expect to need to hold the brake and press the power button when i want to switch to another profile. instead it will switch to another profile even without holding the brake which makes using the dashboard clunky as you need to switch back profile everytime you trigger it by accident

### Contact Information

Samybigio

paulederbaus, 2025-02-09:
Does the issue still exist?

Z1mmb0, 2025-04-02:
Tested and not reproducible with pro2 on Utility 2.7/shfw 3.10.1. 

I assume this guy had a too agressive hall sensor setting.

Samybigio, 2025-04-27:
i apologize for taking so long to reply, first of all, i mentioned that pro2 isnt affected by this issue, which  explains why you werent able to reproduce it. also i dont know if i made mistakes when i was writing the issue but now it only happens with "brake and main button press" which also rules out the aggressive hall sensor setting, just to be sure i tried a much higher value and even although i needed to press a lot to activate the double press one, the single press one just did it without pressing the brake. the issue persists with 3.10.1 (so latest at the time of writing)

oadq, 2025-06-12:
Should be fixed in the next update.

closed 2025-06-23 by oadq