scooterhacking/scooterhacking-issues-public #130: Feature: Enable custom profile switch on speed > 0

Z1mmb0, 2024-10-29:
### Scooterhacking Tool

Firmware (SHFW)

### Type of feature

Other

### Priority

low (take your time)

### SHUT Version

2.6.1

### SHFW Version

3.9.1

### Scooter model

Xiaomi Pro, Pro2, 1s, essential, 3

### Description of the new feature

First of all thank you for implementing the custom profile switch combinations. 

For xiaomi pro2 the switch is not working while driving (speed > 0). 

The desired usecase is to easily switch to a 'legit' profile while driving. 
Example for used combinations: double tap break, double tap accel, 1x tap break. 

If it is possible to implement it in this way, also consider writing a disclaimer that this feature works while driving and combinations should be used with care. 

### Contact Information

dc: z1mmb0

KingBr0, 2024-10-29:
You can just enable one of the presets to act as a way to enter a "legal" mode, since those are quick and easy and work while driving. The custom combination is only while stationary to have a secure way to enable the "illegal" mode so no one can find out it even exists.

Z1mmb0, 2024-10-30:
Also doesn't work at least when only throttle and break are involved l but need to verify this.

Z1mmb0, 2024-11-14:
throttle + break works, but inconsistent. anyway i'll close this one

closed 2024-11-14 by Z1mmb0