scooterhacking/scooterhacking-issues-public #131: Feature: Limit Speed for Display

ViSo36, 2024-11-10:
### Scooterhacking Tool

Firmware (SHFW)

### Type of feature

UI / UX

### Priority

low (take your time)

### SHUT Version

SHUT

### SHFW Version

SHFW

### Scooter model

Ninebot G2

### Description of the new feature

Feature like DMS for Navee Scooters that limits the max speed displayed, so do not incresses sfter certain speed (customizable by user?), even if real speed is higher. 
For example riding at +30kmh but display stops at 25kmh.

### Contact Information

ViSo36

Nyami11, 2024-11-10:
I see it pretty stupid way to aviod police, especially since you can just switch to legal profile by simply brake+throttle pull and actually limiting the speed.

KingBr0, 2024-11-14:
You could just set a lower tyre size (f.e. instead of 10 inches to 6), This would get applied to every profile tho. But as Nyami said just switch to the legal profile with a really simple combination.

closed 2025-06-23 by oadq