scooterhacking/scooterhacking-issues-public #133: BUG: Ninebot g30 gen 4 motor throttle bug

Neros45, 2024-12-17:
### Scooterhacking Tool

Firmware (SHFW)

### Type of issue

Performance issue

### Severity of issue

medium (affects usability but has a workaround)

### SHUT Version

2.6.1

### SHFW Version

3.9.1

### Scooter model

Ninebot G30

### Description of the Bug

When accelerating on 34-35 kmh the scooter doesnt accelerate further until you press like 40 percent of the throttle then you will feel it start accelerating/pulling you when i go over 35 kmh and press full throttle again i feel the scooter slowing me down its feeling the same as i would just let go of the throtthle or if you drive 22kmh and go downhill the scooter starts to block the speed. Many people especially the german people have this problem. This is only a problem for gen4 motors 

### Steps to Reproduce

1. take your scooter and drive till 35kmh
2. you wont get more than 35 kmh
3. press the throttle 40 percent it will go over

### Expected vs. Actual Behavior

Expected: not having to let go of full throttle to go over 35kmh 
Reality: having to go of full throttle 

### Contact Information

neros89

Nyami11, 2024-12-17:
What max power are you using on the curve?

Neros45, 2024-12-17:
> What max power are you using on the curve?

i think you mean the speed curve i use 25a