scooterhacking/scooterhacking-issues-public #135: Feature: Translations

Sencaid, 2025-01-01:
### Scooterhacking Tool

Utility (SHUT)

### Type of feature

UI / UX

### Priority

medium (upcoming releases)

### SHUT Version

Idk

### SHFW Version

Idk

### Scooter model

Ninebot G30

### Description of the new feature

Like discussed in Discord in Channel SH Utility. 
A Function to implement for allowing Translations. These Translations can be made by the Users to contribute. I would like to do German.

### Contact Information

sencait (discord)

nopbxlr, 2025-03-17:
I worked on internalization of an app I was just creating and thought of this ticket - so coming back to it.
Unfortunately, this won't be done for the following reasons:
- the app carries too many hard coded strings and extracting them would take too much time. The effort of moderating and coordinating crowd-sourced translations is also a heavy time consuming process. The effort is worth being put on improving the support of current and of new vehicles for now.
- The technical knowledge required to correctly understand the parameters goes hand in hand with the basic English skills required to use the app.
- To ensure the written community guides match precisely, it is betrer to stick with a universal language. 

Knowing this, I will be closing off this issue, without locking the conversation. Feel free to give opinions on all of this and leads for mitigating the issues I've mentioned if you think of any.
Thank you!

closed 2025-03-17 by nopbxlr

Sencaid, 2025-03-17:
> * Die App enthält zu viele fest codierte Zeichenfolgen, deren Extraktion zu zeitaufwändig wäre. Auch die Moderation und Koordination von Crowdsourcing-Übersetzungen ist sehr zeitintensiv. Der Aufwand lohnt sich daher, um die Unterstützung aktueller und neuer Fahrzeuge zu verbessern.

Do you know how i can do this? i would take the time

BastelPichi, 2025-03-17:
Youd need Source Code access to utility. And this is only provided to trusted, long standing members of the communtiy, usually. Not just some random guy on Github.

I dont think the fact you beeing banned from the ScooterHacking Discord and Telegram, you trying to distribute an modified, translated version of SHU without developer consent or the fact that you are using a translator for this very conversation increases your trustworthiness at all. In fact, it leaves you pretty much with no trustworthiness at all.

nopbxlr, 2025-03-17:
I did not realize they were banned from our communities - you have a very attentive eye