scooterhacking/scooterhacking-issues-public #140: Feature: Add ZT3 support to SHUT

encryptize, 2025-01-20:
### Scooterhacking Tool

Utility (SHUT)

### Type of feature

Communication

### Priority

low (take your time)

### SHUT Version

SHUT 2.6.1

### SHFW Version

x

### Scooter model

Ninebot G30

### Description of the new feature

Adding support for ZT3 now will make it easier to add more models later (e.g., the upcoming G3, F3, E3 models). The most important technical details that complicate things a bit:

- The old BLE target 0x21 is now 0x04.
- The DRV (CTRL) target 0x22 is no longer there. The MCU (Motor controller) with ID 0x02 and the VCU (Vehicle controller) with 0x16 are now in use. For both of these modules, we need to be able to check CPUID.
- The main BMS has ID 0x07, BMS2 most likely for the additional battery has ID 0x06.
- There is a TFT module with ID 0x23, however I don't know if it is of any relevance.

Other useful information:
Manufacturer data: `0x4E430100020000FC`

### Contact Information

dsc: Encryptize

BastelPichi, 2025-01-20:
The BLE now actually supports larger frames, meaning more than 20 bytes of data. Implementing this would be interesting especially for OTA.

Davitton, 2025-05-10:
How can I reset a zt3 pro d back to factory settings? I changed the region and wanted to undo it, but now I can't get rid of the immobilizer and I can't connect either.

BastelPichi, 2025-05-10:
This is not the right place for this.

vegasbillsfan82, 2025-05-25:
i have same issue

RenegadeOG77, 2025-05-26:
Hi my ZT3 will Not be found on SHU how can I Connect it?