scooterhacking/scooterhacking-issues-public #143: Feature: Switching back to the old indicators

IZorixI, 2025-02-04:
### Scooterhacking Tool

Utility (SHUT)

### Type of feature

Communication

### Priority

low (take your time)

### SHUT Version

Upcoming version

### SHFW Version

Upcoming version

### Scooter model

Ninebot G2

### Description of the new feature

Add a switch to the app to have the indicators like in the old firmware. So no hazard lights and therefore no more delay when pressing the buttons.

As an additional option (not important):
Introduce a switch that allows you to adjust the speed of the flashing lights.

That would be great.
Thank you for everything ScooterHacking♥️

IZorixI, 2025-02-04:
I forgot to say my username on Discord. Its: zorix_zrx

BastelPichi, 2025-02-04:
Im pretty sure this is BLE only. SHFW is installed on your DRV. If you downgrade your BLE via Flash -> From Repo -> BLE you should get the old behaviour.

closed 2025-06-23 by oadq