scooterhacking/scooterhacking-issues-public #144: BUG: No power metrics on dash

milankkk, 2025-03-09:
### Scooterhacking Tool

Firmware (SHFW)

### Type of issue

Other

### Severity of issue

low (asthetic, minor issue)

### SHUT Version

2.7(73)

### SHFW Version

3.10.1

### Scooter model

Ninebot G2

### Description of the Bug

When enabling any type of power metrics(like current or wattage), the dash reports 0. As a result, nothing displays on alternating dash data if these are enabled on it.

### Steps to Reproduce

1. Enable Current or Power on main dash data.
2. Drive around (value remains at 0).

### Expected vs. Actual Behavior

Should report amperage or wattage when driving around.

### Contact Information

milan.be

oadq, 2025-06-23:
@milankkk Do you happen to use BMS emulation?

oadq, 2026-01-13:
Fixed in SHFW v3.10.3.