scooterhacking/scooterhacking-issues-public #146: Feature: Make current DPC settings visible

Z1mmb0, 2025-04-02:
### Scooterhacking Tool

Utility (SHUT)

### Type of feature

UI / UX

### Priority

low (take your time)

### SHUT Version

2.7

### SHFW Version

3.10.1

### Scooter model

Xiaomi Pro, Pro2, 1s, essential, 3

### Description of the new feature

We are currently on utility 2.7 and firmware 3.10.1

When dpc curve is set, the UI makes it's hard to know what values were actually set. You just see the curve and no parameters. If you visit your settings after some time you have to roughly estimate e.g. if ampere settings were at 25 or maybe 27 Amps.

There should be a box below the graph which documents the parameters of the currently visible curve. 

### Contact Information

Z1mmb0

reuzurq, 2025-04-02:
You can change to "Advanced" curve builder and see exact mA value per step.

Z1mmb0, 2025-04-03:
Thanks, didn't realize there is actually a slider. 
Also wouldn't expect to have to go advanced at all to see which values are set. 
At least there is a method. closing this one.

closed 2025-04-03 by Z1mmb0