scooterhacking/scooterhacking-issues-public #147: BUG: cc trues to kill the motor

Measupra, 2025-04-11:
### Scooterhacking Tool

Firmware (SHFW)

### Type of issue

Crash

### Severity of issue

medium (affects usability but has a workaround)

### SHUT Version

2.7

### SHFW Version

.2.8

### Scooter model

Xiaomi Pro, Pro2, 1s, essential, 3

### Description of the Bug

When i ride the scooter, cc enables and then it tries to kill the motor by braking rapidly from 43 to 30 and then goes back to 43 and does this repeatedly.

### Steps to Reproduce

1turned on speed based cc

### Expected vs. Actual Behavior

To stay at the same speed just like stock, instead brakes insanely fast which can damage the motor

### Contact Information

Measupra

NormalHuman-Anything, 2025-04-11:
has anyone had same issue like him? never experienced same issue across 4 different scoot models(esx, m365, g30, g2)

oadq, 2025-06-24:
@Measupra Please post all of your cruise control settings.

closed 2026-02-14 by oadq