scooterhacking/scooterhacking-issues-public #149: BUG: Control behavior!!!

Viktor-Kermit, 2025-04-23:
### Scooterhacking Tool

Firmware (SHFW)

### Type of issue

Performance issue

### Severity of issue

high (major functionality broken, no workaround)

### SHUT Version

2.7

### SHFW Version

3.10.1 and prerelease v_9.8

### Scooter model

Ninebot G2

### Description of the Bug

F2 base model (16s!!!)
~30kmh when releasing the throttle it continues to accelerate for about half second or more if headlight is turned or there is more load its worse and its not only at this speed it varies
same with brake but at ~15kmh 
tested g2 ble (1.11.0) but worse at f2 ble (5.6.6) drv`s are all the same no changes (3.10.1 and v_9_8)
happens only on 16s on 10s it`s barely noticeable on 10s 1.5kw peak on 16s 2.7kw because of more power its more noticable (my supposings)
without ovm its worse!

fw start speed- 45 and around this speed with the slightest brake the motor does fast cycles of cuts only around this speed if i brake more its just faster and worse

settings
45a curve (25doesnt fell right)
acc boost-100%
ovm-on

min brake speed-3kmh
curve 43A 
overshoot-60A
no brake boost and autobraking

0 start speed no MSP

fw start speed-45kmh
initial field-5A
Variable-2000mA kmh
70A torque current
80A Field current

Sorry for my bad English.

### Steps to Reproduce

BLE Firmware 

### Expected vs. Actual Behavior

Release brake lever stop braking 
Release Throttle stop accelerating
...
^^^

### Contact Information

Kermit [16S F2]

NormalHuman-Anything, 2025-04-23:
can you check if it's adc reading problem?

Viktor-Kermit, 2025-04-23:
how to check
on 10s its perfect barely noticable with headlight and load

Viktor-Kermit, 2025-04-23:
They are normal 38 in position and go up how they need.

NormalHuman-Anything, 2025-04-23:
maybe at this point could be just field weakening behavior

Viktor-Kermit, 2025-04-23:
it isn`t its with every settings like that tried without it but and even about brake at 45kmh
at this point SHFW 4.0 ...