scooterhacking/scooterhacking-issues-public #150: BUG: Headlight Flashes with "Always on"

xTopher1049, 2025-04-26:
### Scooterhacking Tool

Firmware (SHFW)

### Type of issue

Performance issue

### Severity of issue

medium (affects usability but has a workaround)

### SHUT Version

SHUT 2.7

### SHFW Version

3.10.1

### Scooter model

Ninebot Es2, Es3, Es4

### Description of the Bug

When enabling the light setting "Always on headlight", the headlight flashes on and off indefinitely. 

Turning off the setting and manually turning the headlight on works normally.

### Steps to Reproduce

1. In SHU turn on always on headlight

### Expected vs. Actual Behavior

Headlight is on constant.

oadq, 2025-06-12:
Should be fixed in the next update.

xTopher1049, 2025-09-20:
Hi @oadq , I ask this sincerely. Is there an ETA for next release that could fix this?

oadq, 2025-09-21:
This is fixed in the latest pre-release build, which is not public. I do not know when it will be released publicly.