scooterhacking/scooterhacking-issues-public #151: Feature: Display color changing

wesolekk, 2025-05-10:
### Scooterhacking Tool

Firmware (SHFW)

### Type of feature

Other

### Priority

low (take your time)

### SHUT Version

SHUT 2.7

### SHFW Version

3.10.1

### Scooter model

Xiaomi Pro, Pro2, 1s, essential, 3

### Description of the new feature

Could yall somehow add a feature where i can change the color of the display on my pro 2? Ive seen someone that had a red display.

### Contact Information

wesolek48 (discord)

lekrsu, 2025-05-10:
That needs Camible (ble) firmware and not inherently related to shfw (drv)

sweetpalma, 2025-05-16:
@lekrsu is it the same for Ninebot Max G2?

lekrsu, 2025-05-16:
@sweetpalma There's no **custom** BLE firmware for G2/F2, would have to use existing features of display.

sweetpalma, 2025-05-16:
@lekrsu I believe that some API should already exist - for example, you could enable boost for sport mode in application, and it will make that first bar red. Is there any way to help you to determine that?