scooterhacking/scooterhacking-issues-public #153: BUG: BMS cut off?

betamoon1, 2025-06-05:
### Scooterhacking Tool

Firmware (SHFW)

### Type of issue

Performance issue

### Severity of issue

high (major functionality broken, no workaround)

### SHUT Version

2.7

### SHFW Version

3.10.1

### Scooter model

Ninebot G2

### Description of the Bug

I have set the scooter to speed-based 25A, I recently rode up a hill and the G2 suddenly cut out abruptly. It could no longer be switched on either. After a few seconds, the dashboard beeped, then it started up again. BMS cut perhaps?

### Steps to Reproduce

Drive uphill more slowly than at top speed

### Expected vs. Actual Behavior

Expected: Slow down

Actual: BMS cut off?

Z1mmb0, 2025-06-05:
similar issue:

https://github.com/scooterhacking/scooterhacking-issues-public/issues/141

was closed as a non-firmware related issue.
i suggest share your shfw settings. 

blind guess: turn off accel boost or check your hardware

betamoon1, 2025-06-06:
Hardware runs flawlessly under original firmware, scooter is only 2 weeks old, shfw experience has also been around for a long time since the 1s then g30 2. acceleration boost was never activated and field weakening was also off, so 

speed-based
22kmh
25A

Brake:
45A 

The rest is set as standard

betamoon1, 2025-06-09:
I may have expressed myself incorrectly. It has happened once so far, up a steep hill and once where I let it idle at 30 and braked (curve 52,53,54,55A before). My brake is set with a curve of 42, 43, 44 and then 45 A(specs from german forum). I have now changed to dpc and apart from that it runs great. With DPC, it currently draws a maximum of 40 A, although I have set 25A to approx. 1300 W. Is that normal that it goes over the 25A? Otherwise it runs wonderfully before and after with SHFW.

betamoon1, 2025-06-10:
So update: I have the same settings as before only this time with DPC 25 A curve 0.6 quadratic. I drove up the same hill. Maximum load was 40 A and 1300 W approx. I got up normally and the engine wasn't really warm or hot either, apparently it was speed based.

paulederbaus, 2025-06-26:
@betamoon1 This is likely a bug of sports mode in SHFW for G2 / F2. If this issue still exists, please try the same settings on Drive instead of Sports. 
Also this is very likely related to acceleration and brake boost.

We are looking into the BMS cut off limits and will try to fix this.

betamoon1, 2025-06-26:
@paulederbaus thanks, the problem hasn't happened since i switched to dpc, not even with the same mountain, of course a fix would still be an advantage. thanks for your work!