scooterhacking/scooterhacking-issues-public #158: BUG:  G2 -  wrong Ride Distance

KenshiHH, 2025-07-06:
### Scooterhacking Tool

Firmware (SHFW)

### Type of issue

UI / UX fix

### Severity of issue

low (asthetic, minor issue)

### SHUT Version

2.7

### SHFW Version

3.10.1

### Scooter model

Ninebot G2

### Description of the Bug

The distance appears to be bugged. In the Segway app, the "This Ride" distance starts with a seemingly random value, sometimes even negative.

When you subtract the starting distance from the final distance after a ride, the result is also incorrect.
For example, a 10 km ride might show as adding 100 km.

The same issue occurs when using apps like M365 Tools, the ride starts at 0, but the distance shown after the ride is wrong.

### Steps to Reproduce

1. Start the Scooter
2. Check the Segway App "This Ride" distance

### Expected vs. Actual Behavior

See the actual distance

### Contact Information

kenshihh

ayca10sedo-crypto, 2025-12-03:
> ### Scooterhacking Tool
> Firmware (SHFW)
> 
> ### Type of issue
> UI / UX fix
> 
> ### Severity of issue
> low (asthetic, minor issue)
> 
> ### SHUT Version
> 2.7
> 
> ### SHFW Version
> 3.10.1
> 
> ### Scooter model
> Ninebot G2
> 
> ### Description of the Bug
> The distance appears to be bugged. In the Segway app, the "This Ride" distance starts with a seemingly random value, sometimes even negative.
> 
> When you subtract the starting distance from the final distance after a ride, the result is also incorrect. For example, a 10 km ride might show as adding 100 km.
> 
> The same issue occurs when using apps like M365 Tools, the ride starts at 0, but the distance shown after the ride is wrong.
> 
> <img alt="Image" width="460" height="491" src="https://private-user-images.githubusercontent.com/11234619/462935398-1a11e204-2195-409c-913d-114eb12913fd.png?jwt=eyJ0eXAiOiJKV1QiLCJhbGciOiJIUzI1NiJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NjQ3OTY5NTcsIm5iZiI6MTc2NDc5NjY1NywicGF0aCI6Ii8xMTIzNDYxOS80NjI5MzUzOTgtMWExMWUyMDQtMjE5NS00MDljLTkxM2QtMTE0ZWIxMjkxM2ZkLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTEyMDMlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUxMjAzVDIxMTczN1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTJmM2IyYTBhNGNhYjA3ZDJmMmU3NGE3YzcyMzk4NGUzMWJhYmRhMzViOWUzMmY0YjBkMjAxMzNmYzZiNmIxZmYmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.vl6_UvEq9IdZUfGmThYGp2PYhQ87rhQscavjaMvmjaU">
> ### Steps to Reproduce
> 1. Start the Scooter
> 2. Check the Segway App "This Ride" distance
> 
> ### Expected vs. Actual Behavior
> See the actual distance
> 
> ### Contact Information
> kenshihh

laszlomorhardt6-cmd, 2026-10-27:
Nem megy 35kmes sebességel